OpenChrom/openchrom #548: Agilent OpenLab reader and export to Allotrope Simple model fails

haeussma, 2024-10-08:
First of all, thank you for developing OpenChrom. It is an important tool for harmonizing data from various vendors!

I tried to read in an HPLC Agilent `*.dx` file using the Agilent OpenLab GC-FID converter. Although the reader states it is for GC, it also works for HPLC data, and I could load the chromatogram. I have tried exporting the chromatogram and the processed peak data in the Allotrope Simple Model (ASM) format. I could only select the `ASM GC-MS Chromatogram` converter, although the data is from HPLC. 
I suppose that this issue originates from the reader, who states that it is GC data, although it is not. Thus, the conversion fails.

Do you have any idea how to make this work?

I attached a sample .dx file.
[2024-07-13 11-35-51+02-00-02.dx.zip](https://github.com/user-attachments/files/17294117/2024-07-13.11-35-51%2B02-00-02.dx.zip)

Mailaender, 2024-10-08:
For your `2024-07-13 11-35-51+02-00-02.dx` GC-FID file, please use `ASM GC-FID Chromatogram` to save it instead.
[image]
On Friday there will be a new version which also includes GC-MS export based on the Allotrope Q3/2024 release. The current experimental one was used to provide test data but is not fully compliant.

Mailaender, 2024-10-08:
I might have misunderstood this. The real bug is probably in our OpenLab import converter, which does not guess chromatography and detector type correctly. ASM then not offering the correct choice is just a symptom of it.

haeussma, 2024-10-08:
Yes, this was my suspicion.

Mailaender, 2024-10-10:
Which instrument series was this measured on?

haeussma, 2024-10-11:
Agilent 1260 Infinity 2 controlled by OpenLab 2.1

closed 2024-10-18 by eselmeister