OpenChrom/openchrom #592: Error: "The following connectors are not available: Agilent OpenLab HPLC-UV/Vis"

Lorentary, 2025-01-24:
Hi,

I was trying to use OpenChrom to analyse the HPLC-DAD data we have from Agilent OpenLab CDS. The results are of the .dx format. However, that specific format seems to not be available currently.

Is this caused be an outdated/depricated plugin, a corruption of the source file at converters.openchrom.net, or something else - The error seems very generic, so I cannot tell whether this issue will have a fix in the future or not

Mailaender, 2025-01-24:
We currently only support GC-FID OpenLab `*.dx` files, but if you are willing to share some data we can add that to the converter portfolio. If you don't want to do this in public, use the https://lablicate.com/about/contact form.

Mailaender, 2025-01-24:
I just pushed a new version of OpenChrom. As we always overwrite the past one which is a bit unfortunate, you might need to go to https://www.openchrom.net/download install again and then download the missing plug-in.

prinsalaric, 2025-04-24:
Hi there,

I'm currently using OpenChrom version 1.5.6 and I'm experiencing the same issue with the Agilent OpenLab HPLC-UV/Vis plugin (.dx format). No futher information about the error either. I can confirm that same issue occurs on version 1.5.4.

eselmeister, 2025-04-24:
@prinsalaric Could you provide some files for testing purposes?

prinsalaric, 2025-04-24:
[querc_standards.zip](https://github.com/user-attachments/files/19890282/querc_standards.zip)

Hi @eselmeister , attached are four standards from a run on the instrument.

eselmeister, 2025-04-24:
@prinsalaric Do you expect that your data looks like this?

[image]

Have you installed the OpenLab and ChemStation GC-FID converter?

Mailaender, 2025-04-24:
The HPLC-DAD converter is indeed not installable.

[image]

closed 2025-04-24 by eselmeister

Mailaender, 2025-04-24:
This will be fixed in 1.5.7 which has an ETA of this Friday even though we are impacted by https://www.eclipsestatus.io/incidents

prinsalaric, 2025-04-25:
Hi there @Mailaender and @eselmeister 

Can confirm that the update fixed the issues and I can successfully import plugin and open chromatograms. Much appreciated! :-)